Vertabreak/Marlin #234: [BUG] freeze when changing extruder

88mich, 2020-04-17:
when the extruder changes during printing the printer stops, I noticed that it stops when it encounters the T1 command

[image]

Printer geeetech A10M gt2560 v4.0

Vertabreak, 2020-04-17:
File a bug report to marlin none of that code was written by me.

closed 2020-04-17 by Vertabreak

88mich, 2020-04-17:
sorry! ok thank you!

Vertabreak, 2020-04-17:
he literally documented coding it on his channel to.
https://www.youtube.com/watch?v=SispDLjKXLI

Vertabreak, 2020-10-02:
i have coded complete sets of vtools to be loaded at boot in my fork this should no longer be possible as you cant switch to a tool that does not exist.