Vertabreak/Marlin #235: [BUG] V-tool 1 defaults to 0:0%

iatwalrus, 2020-04-25:
When attempting to print not using virtual extruders in GCode, but using tool change references (T0, T1, etc.), the printer will lockup when selecting T1 as the extruder to print from. I noticed in the Mixer menu, if you set Active V-Tool to 1, the mix is set to 0:0%, and this needs to default to 0:100%. It can be manually adjusted and saved, but the default should be edited. All other tools have an appropriate default percentage. This way when the firmware is installed, manual configuration of this setting is not required and it will work using tool references in GCode instead of virtual

Vertabreak, 2020-04-25:
this is a configuration hosted here file a report with marlin directly.
https://github.com/MarlinFirmware/Marlin/issues

closed 2020-04-25 by Vertabreak

iatwalrus, 2020-04-25:
I have posted it there as well. I just wanted you to know since I am using your fork of Marlin on my A10M.

Vertabreak, 2020-04-25:
only reason there is a fork at all is they wont accept my unified config that isn't how they do things they accepted well over 100k lines from me of everything else in there format. i made sure every geeetech printer was supported my marlin source it self. this fork is maintained only for my personal use anymore.

Vertabreak, 2020-10-02:
i have coded a work around that loads a complete set of 8 vtools in my fork this should no longer be an issue. 
this was more of a user error situation then a bug but i got real tired of hearing about it and nobody reads the wiki or understanding they had to create the vtools.
so now a default set is loaded on bootup and i should no longer hear anything about this.