Vertabreak/Marlin #236: Support GTM32 Pro VD

dasch3a, 2020-04-29:
Hi,

I saw that your Software supports more Geeetech models. Is there also support for the GTM32 Pro VD? I did not see it in the src.

Greetings

David

Vertabreak, 2020-05-01:
the VB and VD are the same as far as marlin gtm32 support is development only does not print.

closed 2020-05-01 by Vertabreak

dasch3a, 2020-05-07:
So I could just choose the GTM32 VB instead of my GTM32 VD and it should be running? Or what do you want to say with "only does not print"?

Vertabreak, 2020-05-08:
none of the gtm32 board are print ready in marlin its unfinished the pins file for VB covers nearly every gtm32 board there is no need to have one copys that have each new name where geeetech made a s tiny change and called it something new.