Vertabreak/Marlin #267: G29 J  resets printer

n00b42, 2020-06-25:
### Description

`G29 J` resets the printer

### Steps to Reproduce

~~~
G28
G29 A
G29 L1
G29 J
~~~

**Expected behavior:**

Tilting 3 points.

**Actual behavior:**

~~~
Send: G29 J
Recv: //action:notification Doing G29
Recv: Tilting mesh (1/3)
Recv:  T:26.50 /0.00 B:25.48 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:26.50 /0.00 B:25.48 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:26.56 /0.00 B:25.45 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:start
Recv: Marlin Marlin 2 Bugfix Build 495
~~~

#### Additional Information

Config: [Marlin.zip](https://github.com/Vertabreak/Marlin/files/4833324/Marlin.zip)

Seems related to: https://github.com/MarlinFirmware/Marlin/issues/14575

n00b42, 2020-06-25:
Should I report this issue upstream or could it be related to this fork?

Vertabreak, 2020-06-25:
nothing to do with my configuration and all of my code is upstream anyways this is a marlin issue this fork is a flavor of config nothing more.

closed 2020-06-25 by Vertabreak