Vertabreak/Marlin #351: Geeetech A10M filament change no work well E1 E2 problem

kondorzs, 2020-09-24:
Hi I upgraded my A10M printer new firmware but I think Filament change and load and unload no work good. I Like chage E0 (E1) put work. Like change (Load E1..(E2) put and load E1 filament again. And I repet metod and load E2. So If put E1menu--> load load E1 and Put E2 menu --> load E1 and put E1menu  again and load E1 .

Vertabreak, 2020-09-28:
this is user error due to lack of understanding of vtools furthermore the config only maintained on this repo bugs must be filed with marlin.

closed 2020-09-28 by Vertabreak

kondorzs, 2020-09-28:
So My problem no problem? Work god If just one extruder work If put menu E1 and E2?

Vertabreak, 2020-10-02:
you must activate the correct Vtool before selecting filament change failing to do so will only do E0. 
once the Vtool for 0%/100% is active rather then 100%/0% filament change works. 
again this is user error due to lack of understanding suggest reading the marlin wiki and fully understanding how vtools work.