Vertabreak/Marlin #362: G29 Autolevel not work

przepka1985, 2020-10-01:
My printer
Geeetech A10T
TMC2208STD All
Genuine BL Touch
Runout Sensor
3 Extruders
MixT

### Description

G29 command dont work from terminal and from start gcode.

### Steps to Reproduce

1. I have check previous relise from 18.09.2020 and autolevel working fine
2. After Verts last clean omething its missing in autolevel sector or bl tuch sector
3. Im back now to build 498 before 18.09.2020

**Expected behavior:** Autolevel works

**Actual behavior:** Short beep from buzer (like click more than beep) And not move at all, on terminal i got an   echo:debug:info, errors    
but without any description

#### Additional Information

[Config.zip](https://github.com/Vertabreak/Marlin/files/5312751/Config.zip)

Vertabreak, 2020-10-02:
G29 isn't a valid command UBL does not work this way zero start gcode or any kind is needed when used correctly.
This is 100% Pebkac error suggest reading the wiki.
https://marlinfw.org/docs/gcode/G029-ubl.html

closed 2020-10-02 by Vertabreak

przepka1985, 2020-10-02:
Thats make sense :) 
I was use to 1.1.8 with bi linear.
Thanks Vert