Vertabreak/Marlin #585: [BUG] Geeetech A20T Case Light on HE2.

Verustus, 2023-03-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When I want to add case light on pin 2 (connected to unused HE2 connector) the light does work but it only lights up on 100% and anything below that doesn't work. I have commented out `#define HEATER_2_PIN` in pins_GT2560_V3.h. I have tried it on pin 6 (connected to LED connector) and it works but I will use more leds and that small transistor wouldn't handle that.

All the edits made by me can by find with F3 by typing "// edited by Verustus"

### Expected behavior

I should be able to controll the brightness of the light from 1% to 100%.

### Actual behavior

It doesn't light up when I set the brightness anywhere from 0% to 99% and it only light up on 100%. Same with the onboard led connected to the same pin.

### Steps to Reproduce

1. In Configuration_adb.h uncomment `#define CASE_LIGHT_ENABLE`
2. in pins_GT2560_V3.h commend out `#define HEATER_2_PIN`
3. in pins_GT2560_V3.h change `CASE_LIGHT_PIN` from `6` to `2`
4. connect 24V light to HE2 connector (not needed there is onboard led)
5. compile and upload to board

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Geeetech A20T

### Electronics

Stock electronics

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[edited files.zip](https://github.com/Vertabreak/Marlin/files/11010646/edited.files.zip)

Verustus, 2023-03-20:
And for some reason since I flashed the firmware with case light enabled it gives me so ofen thermal runaway even if there is not a one.

Vertabreak, 2023-04-04:
bug reports should be filed with marlin directly not thru this fork.

closed 2023-04-04 by Vertabreak

marcokitt2000, 2024-04-15:
Hello Vert how are you long time to see. i am still busy with 3d printing on sovol sv07 klipper and anycubic kobra 2 (1st release marlin based) but i am still using my olde geeetech a20t with your latest update firmware works nice and clean. i saw that you are updated on github merge to the latest marlin updates for many printers but not for a20 series. I hope you will merge that to. Thnxx for all the good stuff i like the slow printer and i used orca slicer when not good i used cura and the otherones to. But i like klipper thats wonderfull . i hope you are still in 3d printing its a good nice hobby.